kendryte/k230_linux_sdk #2: [Feature]:  libffi-3.4.6 to get past undefined variable error.

LeonG71, 2024-09-08:
### Feature description

libffi-3.4.4  causes  undefined variable error.

use libffi-3.4.6 to get past undefined variable error.

### Want resolve what problem

When using Xuantie-900-gcc-linux-6.6.0-glibc-x86_64-V2.10.1, 

libffi-3.4.4  causes  implicit declaration error

tmp-libffi-x86_64-pc-linux-gnu/libffi-3.4.3/src/tramp.c: In function 'ffi_tramp_get_temp_file':
tmp-libffi-x86_64-pc-linux-gnu/libffi-3.4.3/src/tramp.c:262:22: error: implicit declaration of function 'open_temp_exec_file' [-Wimplicit-function-declaration]   262 |   tramp_globals.fd = open_temp_exec_file ();

use libffi-3.4.6 to get past   implicit declaration error

MrThanlon, 2024-12-02:
Current buildroot version use libffi-3.4.4

wangjianxin-canaan, 2026-09-17:
The latest SDK uses **libffi-3.4.6**.

closed 2026-09-17 by wangjianxin-canaan