kendryte/k230_linux_sdk #3: [other]:  info on 01studio  K230  wifi and hdmi

LeonG71, 2024-09-09:
### Other

    1.595838] mmc1: SDHCI controller on 91581000.sdhci1 [91581000.sdhci1] using
 ADMA
[    1.624730] sdhci-dwcmshc-kendryte 91581000.sdhci1: card claims to support voltages below defined range

    1.150001] canaan-vo 90840000.vo: probe
[    1.155549] canaan-vo 90840000.vo: background color is ffffff
[    1.161350] canaan-vo 90840000.vo: vth_line line is a
[    1.166563] canaan-drm soc:display-subsystem: bound 90840000.vo (ops 0xffffff
ff80b19c90)
[    1.174694] canaan-drm soc:display-subsystem: failed to bind 90850000.dsi (op
s 0xffffffff80b1a1b0): -19
[    1.184377] canaan-drm soc:display-subsystem: [drm:canaan_drm_bind] *ERROR* F
ailed to bind all components
[    1.194157] canaan-drm soc:display-subsystem: adev bind failed: -19

fn-link   F89FTSM13-W3   wifi   ( contains RTL8189FTV-VB )    on mmc1 
hdmi  socket output is via LT9611 dsi to hdmi on  90850000 ?

Can someone fix the 01studio dts for these ?
(k230-canmv.dts  has LT9611 in it )

LeonG71, 2024-09-12:
I ah see, you already include the wifi driver in this.
output/k230_canmv_01studio_defconfig/build/rtl8189es-eb51e021b0e1b6f94a4b49da3f4ee5c5fb20b715/
is the driver for RTL8189FTV ... 

and  the lt9611 driver is in there too ...

wangjianxin-canaan, 2024-12-16:
The latest version already supports 01studio HDMI, and WiFi is still not supported.

qc-wl, 2024-12-23:
> 我啊明白了，您已经将 wifi 驱动程序包含在其中。 output/k230_canmv_01studio_defconfig/build/rtl8189es-eb51e021b0e1b6f94a4b49da3f4ee5c5fb20b715/ 是 RTL8189FTV 的驱动程序...
> 
> 而且 LT9611 驱动器也在其中......
Is RTL8189FTV drive done? I need him too.

OctopusET, 2025-01-05:
@wangjianxin-canaan I can't get HDMI working on this device. Any requirements?

wangjianxin-canaan, 2025-01-07:
> > 我啊明白了，您已经将 wifi 驱动程序包含在其中。 output/k230_canmv_01studio_defconfig/build/rtl8189es-eb51e021b0e1b6f94a4b49da3f4ee5c5fb20b715/ 是 RTL8189FTV 的驱动程序...
> > 而且 LT9611 驱动器也在其中......
> 
> Is RTL8189FTV drive done? I need him too.

rtl8189fs代码可用使用。

wangjianxin-canaan, 2025-01-08:
> @wangjianxin-canaan I can't get HDMI working on this device. Any requirements? It's LG 4K HDR display

run vglite_drm  command ,and check if the display is ok?

[root@canaan ~ ]#vglite_drm 
red(255) green(1[   49.157147] lt9611_connect_detect 1
28) blue(16) alp[   49.161565] lt9611_connector_get_modes 
ha(255)
[   49.313945] lt9611_connect_detect 1
[   49.317468] lt9611_connector_get_modes 
drm: Found plane_id: 33 connector_id: 48 crtc_id: 46
drm: 1920x1080 (530mm X300 mm) pixel format AR24
DRM subsystem and buffer mapped successfully
buffer phys: 20b00000
[   49.595275] adj_mode hdisplay 1920
[   49.598716] adj_mode vdisplay 1080
[   49.602117] adj_mode clock 148500
[   49.605443] adj_mode crtc_hdisplay 1920
[   49.609283] adj_mode crtc_vdisplay 1080
[   49.613122] adj_mode crtc_clock 148500
[   49.632354] lt9611 1-003b: h_total: 2200
[   49.636325] lt9611 1-003b: hactive: 1920
[   49.640256] lt9611 1-003b: hsync_len: 44
[   49.644185] lt9611 1-003b: hfront_porch: 88
[   49.648373] lt9611 1-003b: hsync_porch: 192
[   49.652561] lt9611 1-003b: v_total: 1125
[   49.656488] lt9611 1-003b: vactive: 1080
[   49.660414] lt9611 1-003b: vsync_len: 5
[   49.664253] lt9611 1-003b: vfront_porch: 4
[   49.668353] lt9611 1-003b: vsync_porch: 41
[   49.786221] lt9611_bridge_enable 
[   50.309533] lt9611 1-003b: video check: hactive_a=1920, hactive_b=0, vactive=1028, v_total=1125, h_total_sysclk=400
display: 0

OctopusET, 2025-10-17:
> > I'm getting similar issue on K230:
> > ```
> > [   11.905198] canaan-drm soc:display-subsystem: Unbalanced pm_runtime_enable!
> > [   11.912862] canaan-vo 90840000.vo: background color is ffffff
> > [   11.918745] canaan-drm soc:display-subsystem: bound 90840000.vo (ops 0xffffffff810c72c8)
> > [   11.926921] canaan-drm soc:display-subsystem: [drm:canaan_drm_bind] *ERROR* Failed to bind all components
> > [   11.936724] platform soc:display-subsystem: deferred probe pending: (reason unknown)
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > I'm using custom kernel with HDMI patch from linux-xuantie-kernel https://github.com/K230-linux/k230-linux-kernel/tree/hdmi
> > What usually causes this issue?
> 
> It might be that the device tree configuration is incorrect. Which development board are you using? The 01Studio development board uses HDMI by default. The corresponding device tree for the 01Studio is
> 
> https://github.com/ruyisdk/linux-xuantie-kernel/blob/linux-6.6.36/arch/riscv/boot/dts/canaan/k230-canmv-01studio.dts

I'm using CanMV-K230-V1.1

wangjianxin-canaan, 2025-10-17:
> I'm using CanMV-K230-V1.1
canmv 1.1 devicetree is:

https://github.com/ruyisdk/linux-xuantie-kernel/blob/linux-6.6.36/arch/riscv/boot/dts/canaan/k230-canmv.dts

`git pull; rm-rf output; make k230_canmv_defconfig;make; ` With a new compilation, the SDK's canmv 1.1 default display is HDMI.

OctopusET, 2025-10-17:
Hmm I'm using that device tree though. and I still get
```
[   11.905198] canaan-drm soc:display-subsystem: Unbalanced pm_runtime_enable!
[   11.912862] canaan-vo 90840000.vo: background color is ffffff
[   11.918745] canaan-drm soc:display-subsystem: bound 90840000.vo (ops 0xffffffff810c72c8)
[   11.926921] canaan-drm soc:display-subsystem: [drm:canaan_drm_bind] *ERROR* Failed to bind all components
[   11.936724] platform soc:display-subsystem: deferred probe pending: (reason unknown)
```
maybe my patch is wrong

wangjianxin-canaan, 2025-10-17:
Please fully use the repository code https://github.com/kendryte/k230_linux_sdk, recompile a CanMV 1.1 version image, and check if the issue persists.
git pull; rm-rf output; make k230_canmv_defconfig;make;

wangjianxin-canaan, 2026-09-14:
Documentation: https://www.kendryte.com/zh/sdkResource/230linux
CI Images: https://download.kendryte.com/k230/release/linux_sdk_images/daily_build/

closed 2026-09-14 by wangjianxin-canaan