kendryte/k230_linux_sdk #4: Difference between k230_sdk and k23”_linux_sdk?

Ben01233210, 2024-09-20:
### Help

I don’t understand for what this is used for and what the advantages are. Also what is with the big core?

wuwentao, 2024-09-23:
@Ben01233210 
please refer to document for the details:
https://developer.canaan-creative.com/document

Ben01233210, 2024-09-23:
Hey, 
That was not really helpful. I read the docs and did not really find any information about the differences between the sdks and most importantly not how to use the big core. I think it would be nice if you would add such docs. But thanks for the help :)

wangjianxin-canaan, 2024-09-23:
> ### Help
> I don’t understand for what this is used for and what the advantages are. Also what is with the big core?

k230_sdk  is double system sdk :  the big core run  the RT-Smart system, and the little core run the Linux  and version is 5.10.4.

k230_linux_sdk: only  big core  run linux system, and version is 6.6.36.little core  is not running anything, and this sdk is sdk is under development, driver support is not perfect.

Ben01233210, 2024-09-23:
Ok great thanks a lot! Now the big core finally supports Linux really cool! Is there a plan to also support the little core in this sdk?

Ben01233210, 2024-10-06:
Is there a plan to support the small core as it is now in the other sdk? I have a lot of code for the small core as it is in the other sdk. But it would be very useful to have Linux on the big core.

wangjianxin-canaan, 2024-12-02:
> @wangjianxin-canaan Is there a plan to support the small core as it is now in the other sdk? I have a lot of code for the small core as it is in the other sdk. But it would be very useful to have Linux on the big core.

The boot_baremetal command in U-Boot allows you to run bare-metal programs，for example ：ext4load mmc 1:1 0x7000000 /${nuttex} && boot_baremetal 0 0x${fileaddr} 0x${filesize};   # read file and run nuttex system on little core。

wangjianxin-canaan, 2026-09-14:
Documentation: https://www.kendryte.com/zh/sdkResource/230linux
CI Images: https://download.kendryte.com/k230/release/linux_sdk_images/daily_build/

closed 2026-09-14 by wangjianxin-canaan