kendryte/k230_linux_sdk #7: [Bug]: Building opencv fails

OctopusET, 2024-11-26:
### What happened

OpenCV build tries to link RISC-V binaries with x86-64(both host, target) linker.

### Reproduction steps

```
make CONF=BPI-CanMV-K230D-Zero_defconfig 
```

### Hardware board

BPI-CanMV-K230D-Zero

### Software version

675c6f61403d2e051e3efa9f7c19bb6bd52d12f2

### Bug frequency

Everytime

### Anything else

```log
[ 39%] Building CXX object modules/core/CMakeFiles/opencv_core.dir/src/va_intel.cpp.o
[ 40%] Linking CXX shared library ../../lib/libopencv_core.so
/opt/toolchain/Xuantie-900-gcc-linux-6.6.0-glibc-x86_64-V2.10.2/bin/../lib/gcc/riscv64-unknown-linux-gnu/10.4.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /lib64/libz.so: error adding symbols: file in wrong format
collect2: error: ld returned 1 exit status
make[5]: *** [modules/core/CMakeFiles/opencv_core.dir/build.make:1552: lib/libopencv_core.so.4.5.4] Error 1
make[4]: *** [CMakeFiles/Makefile2:2162: modules/core/CMakeFiles/opencv_core.dir/all] Error 2
make[3]: *** [Makefile:166: all] Error 2
make[2]: *** [package/pkg-generic.mk:283: /home/user/k230_linux_sdk/output/BPI-CanMV-K230D-Zero_defconfig/build/opencv4-4.5.4/.stamp_built] Error 2
make[1]: *** [Makefile:23: _all] Error 2
make: *** [Makefile:22: buildroot] Error 2
make CONF=BPI-CanMV-K230D-Zero_defconfig -j1  222.47s user 11.97s system 919% cpu 25.498 total
```

MrThanlon, 2024-11-27:
Try docker image `ghcr.io/kendryte/k230_sdk:latest`, mount the toolchain to /opt/toolchain

OctopusET, 2024-11-29:
@MrThanlon Working!
Docker command that I used
```
docker run -u root -it -v $(pwd):$(pwd) -v /opt/toolchain/Xuantie-900-gcc-linux-6.6.0-
glibc-x86_64-V2.10.1/:/opt/toolchain/Xuantie-900-gcc-linux-6.6.0-glibc-x86_64-V2.10.1 -w $(pwd) ghcr.io/kendryte/k230_sdk /bin/bash
```

closed 2024-12-02 by wuwentao