kendryte/k230_linux_sdk #17: [Bug]: K230 canmv v3 hdmi output problem

wilesun5060, 2025-02-11:
### What happened

K230 CANMV V3 is a new board. The current dtl configuration should be wrong. When doing HDMI output, it cannot be output to an external display.

### Reproduction steps

canaan-drm
Encoders:
id      crtc    type    possible crtcs  possible clones
47      0       DSI     0x00000001      0x00000001

Connectors:
id      encoder status          name            size (mm)       modes   encoders
48      0       connected       DSI-1           480x800         1       47
  modes:
        index name refresh (Hz) hdisp hss hse htot vdisp vss vse vtot
  #0 480x800 60.00 480 560 580 600 800 1020 1030 1100 39600 flags: ; type: preferred, driver

### Hardware board

k230 canmv v3 board

wangjianxin-canaan, 2025-02-12:
default display is lcd,Please manually switch the device tree.
cd /boot/；rm -rf k.dtb;ln -s k230-canmv-v3.dtb k.dtb;

[image]

wilesun5060, 2025-02-14:
it is ok,thanks.

closed 2025-02-14 by wilesun5060