kendryte/k230_linux_sdk #23: [help]: BPI的k230D Zero的小板子wifi无法驱动起来，也是8189ftv，用bpi那个选项编译或者canmv230d都不行。

rufle, 2025-03-28:
### Help

[help]: BPI的k230D Zero的小板子wifi无法驱动起来，也是8189ftv，用bpi那个选项编译或者canmv230d都不行。

wangjianxin-canaan, 2025-09-01:
8189驱动没有安装，需要手动安装下 modprobe 8189fs

cctsirjin, 2025-09-04:
> BR2_PACKAGE_WPA_SUPPLICANT wpa相关工具需要配置上。aic8800_bsp 驱动可以不用安装
@wangjianxin-canaan 编译之前`make clean`了一下，WPA工具配置上了。现在`lsmod`里有`8189fs`，WPA工具也能被找到并启用，但是`wlan0`这个interface的flag不知道为何在`ifconfig -a`里没有。

```bash
[root@canaan ~ ]#sta.sh wlan0 MyWifi-2p4GHz MyPassword
```
```text
+ wlandev=wlan0
+ ssid=MyWifi-2p4GHz
+ password=MyPassword
+ set -x
+ cat
+ ps
+ grep wpa_supplicant
+ grep -v grep
+
+ wpa_supplicant -D nl80211 -i wlan0 -c /etc/wpa_supplicant.conf -B
Successfully initialized wpa_supplicant
Could not read interface wlan0 flags: No such device
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Could not read interface wlan0 flags: No such device
wlan0: Failed to initialize driver interface
wlan0: CTRL-EVENT-DSCP-POLICY clear_all
+ sleep 2
+ wpa_cli -i wlan0 scan
Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory
+ sleep 1
+ wpa_cli -i wlan0 scan_result
Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory
+ sleep 1
+ wpa_cli -i wlan0 add_network
Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory
+ wpa_cli -i wlan0 set_network 1 ssid '"MyWifi-2p4GHz"'
Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory
+ wpa_cli -i wlan0 set_network 1 psk '"MyPassword"'
Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory
+ wpa_cli -i wlan0 select_network 1
Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory
+ which udhcpc
+
+ udhcpc -i wlan0 -q -n
udhcpc: SIOCGIFINDEX: No such device
+ wpa_cli -i wlan0 status '//查看网络状态'
Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory
```
```bash
[root@canaan ~ ]#ifconfig -a
```
```text
lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:712 errors:0 dropped:0 overruns:0 frame:0
          TX packets:712 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:52672 (51.4 KiB)  TX bytes:52672 (51.4 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
```

wangjianxin-canaan, 2025-09-05:
sdio0接口默认是关闭的，需要在设备树里面使能sdio0.

output/BPI-CanMV-K230D-Zero_defconfig/build/linux-7d4e1f444f461dbe3833bd99a4640e7b6c2cd529/arch/riscv/boot/dts/canaan/bananapi-canmv-k230d-zero.dts 里面增加如下内容：
`&mmc_sd0 {
	status = "okay";
	no-1-8-v;
	// rx_delay_line = <0x0d>;
	// tx_delay_line = <0xc0>;
};`

cctsirjin, 2025-09-06:
> sdio0接口默认是关闭的，需要在设备树里面使能sdio0.
> 
> output/BPI-CanMV-K230D-Zero_defconfig/build/linux-7d4e1f444f461dbe3833bd99a4640e7b6c2cd529/arch/riscv/boot/dts/canaan/bananapi-canmv-k230d-zero.dts 里面增加如下内容： `&mmc_sd0 { status = "okay"; no-1-8-v; // rx_delay_line = <0x0d>; // tx_delay_line = <0xc0>; };`

按照如上更新设备树dts、重新编译dtb并替换板子`/boot/`目录内的文件后，板子启动后卡在Soft lockup，无法正常加载系统。

过程：

我用的设备树是`output/k230d_canmv_defconfig/build/linux-7d4e1f444f461dbe3833bd99a4640e7b6c2cd529/arch/riscv/boot/dts/canaan/k230d-canmv.dts`，在该文件末尾添加了下面的内容：

```c
&mmc_sd0 {
        status = "okay";
        no-1-8-v;
        // rx_delay_line = <0x0d>;
        // tx_delay_line = <0xc0>;
};
```

然后用`make linux-rebuild`重新编译生成新的`k230d-canmv.dtb`，用这个新的dtb覆盖板子`/boot/`目录下的同名旧文件。（或者`make linux-rebuild`后，再`make uboot-rebuild`，最后`make CONF=k230d_canmv_defconfig`，以下现象相同。）

上电启动，最初可以确认设备树变更得到反映：
```text
# 启动日志看到两者似乎有地址冲突
[    2.580457] mmc1: new high speed SDIO card at address 0001
[    2.608576] mmc0: new high speed SDHC card at address 0001
```
但是之后启动就卡在如下处，循环几次后，UART通信中断，不能正常启动。
```text
# 卡在以下的部分，反复循环
[   36.002791] watchdog: BUG: soft lockup - CPU#0 stuck for 22s! [modprobe:123]
[   36.009856] Modules linked in: 8189fs(O+)
[   36.013871] CPU: 0 PID: 123 Comm: modprobe Tainted: G           O       6.6.36 #4
[   36.021354] Hardware name: Canaan CanMV-K230 (DT)
[   36.026056] epc : _raw_spin_unlock_irqrestore+0x1a/0x2e
[   36.031290]  ra : _raw_spin_unlock_irqrestore+0x16/0x2e
[   36.036516] epc : ffffffff80b29ff6 ra : ffffffff80b29ff2 sp : ffffffc8008c3790
[   36.043737]  gp : ffffffff811e83b8 tp : ffffffd8020cbc00 t0 : ffffffd802125000
[   36.050957]  t1 : 0000000000000002 t2 : ffffffff8d021780 s0 : ffffffc8008c37b0
[   36.058177]  s1 : 0000000000000002 a0 : ffffffd802f99c00 a1 : 0000000200000022
[   36.065397]  a2 : 0000000000000001 a3 : 0000000000000001 a4 : 0000000000000000
[   36.072616]  a5 : 0000000000006857 a6 : 0000000000005144 a7 : 0000000000000000
[   36.079836]  s2 : ffffffc800723690 s3 : 0000000000000000 s4 : 0000000200000022
[   36.087055]  s5 : 0000000000000000 s6 : ffffffc800bc2a58 s7 : ffffffc800bc2a08
[   36.094275]  s8 : ffffffc800bc2c58 s9 : ffffffc800bc2a40 s10: ffffffc800722a58
[   36.101495]  s11: 0000000000000000 t3 : ffffffffffffffff t4 : 0000000000000000
[   36.108714]  t5 : 0000000000000001 t6 : ffffffff811c7458
[   36.114023] status: 0000000200000120 badaddr: 0000000000000000 cause: 8000000000000005
[   36.121937] [<ffffffff80b29ff6>] _raw_spin_unlock_irqrestore+0x1a/0x2e
[   36.128467] [<ffffffff80094008>] __timer_delete_sync+0x6c/0xca
[   36.134304] [<ffffffff80094074>] timer_delete_sync+0xe/0x16
[   36.139880] [<ffffffff012a18a0>] _rtw_free_sta_priv+0x72/0x120 [8189fs]
[   36.146913] [<ffffffff012bb076>] rtw_free_drv_sw+0x9e/0x132 [8189fs]
[   36.153684] [<ffffffff012bb27c>] rtw_drv_add_vir_if+0x136/0x176 [8189fs]
[   36.160801] [<ffffffff012bcae0>] rtw_drv_init+0xa8/0x152 [8189fs]
[   36.167309] [<ffffffff808431f2>] sdio_bus_probe+0xf4/0x110
[   36.172804] [<ffffffff8060d74c>] really_probe+0x84/0x23c
[   36.178120] [<ffffffff8060d960>] __driver_probe_device+0x5c/0xdc
[   36.184126] [<ffffffff8060da0c>] driver_probe_device+0x2c/0xb2
[   36.189960] [<ffffffff8060db8c>] __driver_attach+0x6c/0x11a
[   36.195532] [<ffffffff8060b8d6>] bus_for_each_dev+0x62/0xb0
[   36.201106] [<ffffffff8060d19a>] driver_attach+0x1a/0x22
[   36.206417] [<ffffffff8060caf8>] bus_add_driver+0xce/0x1d6
[   36.211902] [<ffffffff8060e7e2>] driver_register+0x40/0xda
[   36.217388] [<ffffffff80843002>] sdio_register_driver+0x22/0x2a
[   36.223311] [<ffffffff013bd058>] rtw_drv_entry+0x58/0x1000 [8189fs]
[   36.229979] [<ffffffff8000212a>] do_one_initcall+0x56/0x194
[   36.235557] [<ffffffff8008d280>] do_init_module+0x4e/0x1b0
[   36.241046] [<ffffffff8008e9b4>] load_module+0x1368/0x175a
[   36.246533] [<ffffffff8008ef68>] init_module_from_file+0x76/0xae
[   36.252541] [<ffffffff8008f19a>] __riscv_sys_finit_module+0x1c2/0x2a2
[   36.258983] [<ffffffff80b212bc>] do_trap_ecall_u+0x124/0x136
[   36.264645] [<ffffffff80b2a5b8>] ret_from_exception+0x0/0x64
```

wangjianxin-canaan, 2025-09-24:
make linux-menuconfig ;把CMA_SIZE_PERCENTAGE调小一点。

[image]

make linux-rebuild;

wangjianxin-canaan, 2025-09-26:
[image]

cctsirjin, 2025-09-26:
> <img alt="Image" width="1165" height="603" src="https://private-user-images.githubusercontent.com/102512117/494218538-beb5e9e8-f300-439c-ae19-1df6cbead07d.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTg4ODQ5NzQsIm5iZiI6MTc1ODg4NDY3NCwicGF0aCI6Ii8xMDI1MTIxMTcvNDk0MjE4NTM4LWJlYjVlOWU4LWYzMDAtNDM5Yy1hZTE5LTFkZjZjYmVhZDA3ZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwOTI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDkyNlQxMTA0MzRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yMmI2OTkwYjRkNDk0Y2FlYjZjZjcwMmVhNzZlMWI5MmRlMzUxYjI3NmM0ZDg3OGFmZjE3NWYzOGYzMTEwYzQ2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.pP3j-PfxzuJQn4kLJ5ssnIVFrvFKjddFmY8SUvwKtQU">

不好意思，没注意看。现在WiFi能够上电自连了，非常感谢！不过配置文件默认没有启用`fw_setenv`，需要在Buildroot里手动开启一下`Target packages -> Hardware handling -> u-boot tools -> fw_printenv`。

总结一下K230D Zero小板子启用WiFi的整个过程：

- 板子不外接天线的话，存在偶尔连接不上SSH的情况。（我的板子接在凑近路由器的NAS上，信号还行，只有DHCP的IP地址lease到期了会断连。）

- 使能与WiFi模组和功能有关的Linux编译选项：
```bash
make menuconfig CONF=k230d_canmv_defconfig
# 按下`/`可进行搜索
# Target packages -> Hardware handling -> Firmware -> rtl8189fs
# Target packages -> Hardware handling -> u-boot tools -> fw_printenv (fw_setenv is included in this fw_printenv option)
# Target packages -> Networking applications -> wpa_supplicant以及其下全部的包
# Target packages -> System tools -> util-linux -> rfkill
```

- 缩小`CMA_SIZE_PERCENTAGE`：
```bash
make linux-menuconfig
# 按下`/`键搜索`CMA_SIZE_PERCENTAGE`
# Library routines -> DMA Contiguous Memory Allocator (DMA_CMA=y) -> Percentage of total memory (CMA_SIZE_PERCENTAGE [=50])
# 然后将其比例从50调小，比如25。
make linux-savedefconfig
```

- 设备树中添加WiFi的SDIO节点：
```bash
# 可在宿主机环境中编辑
nano output/k230d_canmv_defconfig/build/linux-7d4e1f444f461dbe3833bd99a4640e7b6c2cd529/arch/riscv/boot/dts/canaan/k230d-canmv.dts
```
```C
&mmc_sd0 {
        status = "okay";
        no-1-8-v;
        // rx_delay_line = <0x0d>;
        // tx_delay_line = <0xc0>;
};
```

- 编译修改后的Linux设置并重新生成系统镜像：
```bash
# 在Docker容器中执行
cd ~/k230_linux_sdk
make linux-rebuild
make CONF=k230d_canmv_defconfig
```

- 在板子上，检查和（暂时）安装Linux内核的WiFi模块：
```shell
lsmod
modprobe 8189fs # 安装模块（瑞昱Realtek RTL8189FS）
modprobe -r 8189fs # 卸载模块
```

- 在板子上，启动、关闭以及配置WiFi：
  - 使用`sta.sh`：
https://github.com/kendryte/k230_linux_sdk/tree/a22361b72fc5c33ec68ea10919bf576b6b7c5408?tab=readme-ov-file#how-to-configure-wifi
```shell
# 连接WiFi
sta.sh wlan0 <ssid> <passwd>
# 让U-Boot在上电启动之后、用户login之前就自动连接WiFi。
fw_setenv wlanssid  H3C_wjx
fw_setenv wlanpass 12345678
reboot
```