kendryte/k230_linux_sdk #24: [help]: 如何修改编译kernel以及uboot呢？

Cjiio, 2025-04-12:
### Help

我看kernel和uboot都是buildroot直接编译，从config里找到了kernel的git链接，但是不知道怎么去更方便的修改

wangjianxin-canaan, 2025-04-14:
make uboot-rebuild #rebuild uboot
make uboot-dirclean #uboot clean
make linux-menuconfig #modify configuration
make linux-savedefconfig #save to defconfig

make linux-rebuild  #rebuild linux
make linux-dirclean #linux clean

更多 请 参考https://buildroot.org/downloads/manual/manual.html

Cjiio, 2025-04-15:
> make uboot-rebuild #rebuild uboot make uboot-dirclean #uboot clean make linux-menuconfig #modify configuration make linux-savedefconfig #save to defconfig
> 
> make linux-rebuild #rebuild linux make linux-dirclean #linux clean
> 
> 更多 请 参考https://buildroot.org/downloads/manual/manual.html

但是在sdk里没有找见kernel的文件目录

wangjianxin-canaan, 2025-04-16:
> 但是在sdk里没有找见kernel的文件目录

linux代码使用的仓库是
BR2_LINUX_KERNEL_CUSTOM_REPO_URL="https://github.com/ruyisdk/linux-xuantie-kernel.git"
BR2_LINUX_KERNEL_CUSTOM_REPO_VERSION="7d4e1f444f461dbe3833bd99a4640e7b6c2cd529"
https://github.com/kendryte/k230_linux_sdk/tree/dev/buildroot-overlay/linux  目录下是k230有修改的补丁。

你本地修改linux代码的话，修改类似 output/k230_canmv_01studio_defconfig/build/linux-7d4e1f444f461dbe3833bd99a4640e7b6c2cd529/   目录下的文件就可以了。
修改完后执行   make linux-rebuild ;make 重新编译下就可以了。

Cjiio, 2025-04-16:
> > > make uboot-rebuild #rebuild uboot make uboot-dirclean #uboot clean make linux-menuconfig #modify configuration make linux-savedefconfig #save to defconfig
> > > make linux-rebuild #rebuild linux make linux-dirclean #linux clean
> > > 更多 请 参考https://buildroot.org/downloads/manual/manual.html
> > 
> > 
> > 但是在sdk里没有找见kernel的文件目录
> 
> sdk里面只存放了k230有修改的补丁( https://github.com/kendryte/k230_linux_sdk/tree/dev/buildroot-overlay/linux )
> 
> linux代码使用的仓库(详见配置文件buildroot-overlay/configs)是 BR2_LINUX_KERNEL_CUSTOM_REPO_URL="https://github.com/ruyisdk/linux-xuantie-kernel.git" BR2_LINUX_KERNEL_CUSTOM_REPO_VERSION="7d4e1f444f461dbe3833bd99a4640e7b6c2cd529"
> 
> 你本地修改linux代码的话，修改类似 **output/k230_canmv_01studio_defconfig/build/linux-7d4e1f444f461dbe3833bd99a4640e7b6c2cd529/** 目录下的文件就可以了。 修改完后执行 **make linux-rebuild ;make** 重新编译下就可以了。

了解了，非常感谢

closed 2025-04-16 by Cjiio